JPZV/BluN64-ESP32 #2: Please provide attribution and compatible license

mitchellcairns, 2023-01-28:
Hi, I think what you've done with this is really cool and I love it.

Could you please provide attribution in the credits section for HOJA?

Please also note the license for HOJA is non-commercial share-alike which is incompatible with the license applied to this project. Please update the software license to meet compatibility if you can. This is to prevent people from selling our software without prior arrangements with us.

Thank you

JPZV, 2023-01-28:
Hi!

I thought that I had already added you to the _Special Thanks_. My mistake, I'll do it right away!

About the license, I wasn't aware about that, I'll change it now to CC-NC just like HOJA.

Thank you for letting me know!! Also, nice job with HOJA! It worked completely fine!!! Though, I had to [fork it](https://github.com/JPZV/Blue-N64-Control-ESP32/tree/prev_version) to maintain some features (like connecting automatically to a NSwitch) while you re-implement them!

closed 2023-01-28 by JPZV

mitchellcairns, 2023-01-28:
Thank you, I highly appreciate it!!! <3

Keep up the fantastic work.