JPZV/BluN64-ESP32 #5: Watchdog got triggered

eduardocintas, 2023-05-20:
Hi Bro, fist THANK YOU, and greets to U. It's an awesome project that mix Nostalgia and Nerdiness well.

I'm programmer but new into MicroControllers.(only do some PIC practices back in my teens) 

I'm very interested in your project, and excited to help convert it in somehow "universal Smart Game Controllerr". Started with this cause i bought a cheap cabled fighthstick(Named NIGACN), im manly interested to use it on my TvBox just to discover that most TvBoxes does not include drivers fog cabled GamePads. Then remembered Mundo Yakara's video about "BLUETOOTH ARCADE control" and saw its based in your project but his does not keep updates, so i tried tu use yours. I found a cheap "ESP-WROOM-32"(clone of Wemos Battery) with slot for a 18650 and charger included, so got that and started modding my fightstick. Tried Yakas firmware and worked OK. Manages to get your project to build Flashed with download tool. An worked too.  Then trying to figure out Pinouts in order to get all buttons working i fried my module. i don't really know what i do wrong but i read online tha some GPIOs arre no "safe" as input cause they output at boot time. I will try to figure out to make thing safer to other users.

Long history "short", i got another board. Cause I am in a hurry to work on the project the weekend i got the only one I found locally "ESP32 DEVKITV1" with a "new"(for me) usb to serial chip CH9102X. I started from triying to flash if with download tool and same binaries i built back. I can´t keep geting "[espDownloader.py][line:773][ERROR]: ESP32 Chip efuse check error esp_check_mac_and_efuse".

So i tried flashing Yakaras firmware with webtool and worked like a cham, end flashing it lite blue led and gt recognized as "Blue Gamepad". Then i tried flashig your firmware with webtool it flashed OK, but then keeps rebooting by watchdog wi this messages:
> ```
> E (5498) task_wdt: Task watchdog got triggered. The following tasks/users did not reset the watchdog in time:
> E (5498) task_wdt:  - IDLE (CPU 0)
> E (5498) task_wdt: Tasks currently running:
> E (5498) task_wdt: CPU 0: main
> E (5498) task_wdt: CPU 1: IDLE
> E (5498) task_wdt: Print CPU 0 (current core) backtrace
> ```

Just to test tried you firmware with Yakaras pinout from your web, it installed but don't lite blue led console keeps printing "
I (199632) core_ns_start: Controller already connected" and detected as "Pro Controller"

Tried again with Original pinout butt v2013.04.01(why 2013?) and his time it reboots couple of times cause of watchdog, but it keep on (without blue led) and detected as "Pro Controller".

I will now triple check pins safety and cable my fightstick to the boart to keep testing. If u need any testing i will be glad. If I can fix by myself i will make PR(but im still too noob with this platform).

I feel we have a "PINOUTs HELL" in order to make the project easy to  use for anyone. I'm trying to make an spreadsheet to match controller hardware, modes, and modules/boards pinouts. So we can guess best mapping universal and safe.

IDK if we can auto detect Board/Module and map accordingly, otherwise i think we can make a short user config using safe pins at first boot to define conflicting pins. What u think?

eduardocintas, 2023-05-20:
Update:
Already wired and tested,(With v2013.04.01, v2013.04.02 keeps rebooting by watchdog).

Connected Correctly To Android Phone, tested with gamepad Tester and got:
DPad crossed Right to Left and Up to Down
A, L, R, Z(as R2) and Start = OK, worked as expected
B read as X
C-Dpad , and Home ?Does Nothing

Connected Correctly To Android TvBox, can't get gamepad Tester(tomorrow will install apk):
DPad crossed Right to Left and Up to DownNothing more seems to work

Connected Correctly To Win11 PC, tested with windows app:
DPad crossed Right to Left and Up to Down
Can't see button names but at least aal A, B, X and Y,start and Z seems tu work
C-Dpad Left puts X otation to Min value
C-Dpad Right puts X rotation to Max value
C-Dpad Left puts Y rotation to Min value
C-Dpad Left puts Y rotation to Max value

Also when gamepad disconnects from windows, if i find again it keeps rebooting.

JPZV, 2023-05-23:
Hi, so so so sorry about the late response, I'm being extremely busy lately.

> Then i tried flashig your firmware with webtool it flashed OK, but then keeps rebooting by watchdog wi this messages:

It's so strange, I just tested both 2023.04.01 and 2023.04.02 and both works correctly at least to the point of being connected to the console (I don't have a N64 controller over here so I cannot test it completely). I didn't get any watchdog error or anything else.

The "Controller already connected" it's just a info message (I think I'll change it to as a Debug one), it means that it's waiting for a connection to a console. Remember that, after flashing, you have to go to the Switch Home -> Controllers -> Change Grip/Order and wait until the controller connects to the Switch (it's automatic, you don't have to press L/R).

> and excited to help convert it in somehow "universal Smart Game Controllerr"

Right now this project is just for N64 controllers, specially for the joysticks which are not analog. But on the behind, there is another project called BluControl which I'm working together with Yakara. That project will work with virtually any controller, even self-made ones. You can flash a pre-made firmware or even making your own without modifying the code, just setting up the pinout and that's it. Right now, that project is not public until Yakara release the last video about converting retro controllers to Bluetooth, but I hope it'll be soon.

> i got the only one I found locally "ESP32 DEVKITV1" with a "new"(for me) usb to serial chip CH9102X.

Rather than checking the devkit name, look at the SoC model. It must be printed on the shield (the square-ish metal with a hole) and it should say something like "ESP32 WROOM" or "ESP32-S" or anything like that. With that, check the flash size on the sheets/schematics, it should be at least of 4MB in order to flash this project.

> Also when gamepad disconnects from windows, if i find again it keeps rebooting.

I hate that, It happens to me also and I think it's related on how Windows/PC Bluetooth Drivers handles the gamepad. Not sure if I should fix it or if someone else should, because I also get some strange behavior like explorer.exe being restarted. This also happens on Ubuntu. The only workaround is to unpair and then pair again the gamepad, both in Windows and Ubuntu.

For the main issue, the Watchdog reboot, I would need some more info, like the full serial output (from "start to start". i.e. From the first boot message to the second one), the ESP32 model, and which version you flashed from the web if aplicable.

eduardocintas, 2023-05-23:
> Hi, so so so sorry about the late response, I'm being extremely busy lately.

NVM, many thx for ur time.

> It's so strange, I just tested both 2023.04.01 and 2023.04.02 and both works correctly at least to the point of being connected to the console (I don't have a N64 controller over here so I cannot test it completely). I didn't get any watchdog error or anything else.

As i told, both of my boards have same SOM(ESP-WROOM-32), first one worked "OK" (at least booted and connected) with 2023.04.02 but new one reboot with watchdog message. As first guess i suspected from USB to TTL IC cause i haven't heard from it before. But now i'm pretty sure that new one have a weak regulator. After testing with fist flash tried flash again and got boot loops with any firmware googled error and fond many ppl resolved by changing board 3.3v regulator. I tried feeding 3.3v from external programmer(not connected UART) plus USB and now it can but even 2023.04.02(I think it need a little more power). My first boar(happily it seems not dead), have stronger power cause it handles a 18650 with changing in board.

Still i can't explain why fist board flashed OK with flash download tool v3.9.4. But new one gets "efuse check error", but i found that V3.4.4 flash ok with my new board. So I'm now ready to build and flash. I will try help and provide PRs. At least i can use this project to test and learn and be ready to help with BluControl when code available.
 
> The "Controller already connected" it's just a info message (I think I'll change it to as a Debug one), it means that it's waiting for a connection to a console. Remember that, after flashing, you [...]

Sadly i don't Own a switch, i'm mainly Android and windows user, and interested on BlueGamepad mode.
 
> Right now this project is just for N64 controllers, specially for the joysticks which are not analog. But on the behind, there is another project called BluControl which I'm working together with [...]

I know, but i'm learning with what is public, i will be glad to help u both with coding and/or testing. Do u have ane discord/telegram/whatever group to keep up? i'm Argentinian living on Cali Colombia i can be way more clear in spanish xD.

> Rather than checking the devkit name, look at the SoC model. It must be printed on the shield (the square-ish metal with a hole) and it should say something like "ESP32 WROOM" or "ESP32-S" or [...]

I know(now xD) u actually refer to SoM(Both my boards use "ESP-WROOM-32", SoC is behind the square-ish metal VERY tiny and requires a lot of other components and well design to make it work(that is what modules are for) I think they are out of scope for hobbyists. With SoM's i think the way to go is buy them alone, and get a good quality regulator, and as far i learned some cap to add to Yakaras design to estabilice power on consume peaks.

In the other hand, I think that for projects big enough(Fightsticks, hitbox, arcades, etc), board are the way to go, the are way simpler many times u dont even need to solder. And them resolves many common problems/features. As power regulation, battery management, even soma have fancy oled screens that can look cool.  If we can collect documentation on many of the and take in account exposed pins, and special ones. We can recommend some trouble free options to users. that they can get, flash and connect. 

> I hate that, It happens to me also and I think it's related on how Windows/PC Bluetooth Drivers handles the gamepad. Not sure if I should fix it or if someone else should, because I also get some [...]

I will try to hunt this next weekend if can, so i can learn and help.

> For the main issue, the Watchdog reboot, I would need some more info, like the full serial output (from "start to start". i.e. From the first boot message to the second one), the ESP32 model, and which version you flashed from the web if aplicable.

I have the full log, but i'm running 2023.04.02 now on new board with "dual power", even my own build. So i'm pretty sure it was a power issue. I think the issue can be closed to keep thing clean. But idk where to talk about collaboration.

Have u read my "Update" reporting about Mappings(BlueGampad mode)? Many buttons does not work as documented. I will try to fix that myself. If i can i will send PR

Again many thx for ur work and time.
Best regards.