JPZV/BluN64-ESP32 #6: ESP32/nimble forgets BLE pairing connection to Android client

eduardocintas, 2023-05-28:
I think we are affected by this [ESP/nimble issue](https://github.com/espressif/esp-nimble/issues/16),  it have a [fix](https://github.com/ac-frg/esp-nimble/commit/b1946ea9b4ecc430059359c6eb36d3190da4b301) but is on a branch still not merged. So i don't know hot to set ESP-IDF to use that branch.

Any advice appreciated, i will try patching manually to confirm if resolves out pairing Issues.

JPZV, 2023-05-28:
Nice looking. Although I'm not sure if we're using ESP/Nimble indirectly as we're using [h2zero/esp-nimble-cpp](https://github.com/h2zero/esp-nimble-cpp/). I'll keep an eye on that issue and I'll bump the module once it's completely fixed if needed, as it's an nasty fix right now.

eduardocintas, 2023-05-29:
Tried "nasty fix"(manual patch) and solved Android pairing(So we are using it).

Windows still have same Behavior.
I think we are affected by [this Issue](https://github.com/espressif/esp-idf/issues/5105) 
[This post](https://github.com/espressif/esp-idf/issues/5105#issuecomment-1407540749) describes our proble exactly, i think ist whats happens.

Sadly issue have no updates from February, i will check a related one, i think they moved the discusion there.

eduardocintas, 2023-05-30:
I have tried esp32-ble-gamepad example, and got same windows connection Behavior, so we can discard a problem on our code. pretty sure is a esp-nombre issue.

eduardocintas, 2023-06-02:
Updated all components to latest versiones and got Android pairing Ok without need for Nasty fix.

eduardocintas, 2023-06-02:
Talked too fast, i can now turn bt off and on get reconnected and works OK, can even reboot windows and still get reconnected and works ok. but if i powercicle esp32 i get paired-connected loop same as b4. #$@!!

JPZV, 2023-06-02:
Could you upload the code to a zip or to a repository so I could check what could be wrong?

eduardocintas, 2023-06-03:
[image]

[image]
With Nasty Fix
[image]

[image]

[image]

[image]

Finally i have BñuN64-ESP32 based on Main
[image]

With following changes:
N64-blueretro/main/main.cpp, with BLE-Gamepad sample to easier test and discard aur code as possible culprit
[image]

sdkconfig.defaults changed to lit the led on my board:
[image]

n64-sdk\n64-components\include\n64-pad.h with defines to adjust based on board tyoes
[image]
[image]

n64-sdk\n64-components\Kconfig with changes to be able to select new pinouts on menuconfig:
[image]

n64-blueretro\sdkconfig and n64-switch\sdkconfig have Many changed not done by me. I think they are auto detected at build

I will make a new branch with my code if u like to test.

eduardocintas, 2023-06-03:
Cant create a branch, so forked to do so:
https://github.com/eduardocintas/BluN64-ESP32/tree/Eduardo-Cintas

You can get my code base from there if u want to test. Sould be good to make a discord Channel or telegram for better comunicación . With Victorkte and Yakara i think we are 4 active on the project,