JPZV/BluN64-ESP32 #7: ESP32 problem with button combo

victorkte, 2023-07-09:
Hi JPZV

I'm atempting to do the mod again but I ran into problems again

I flashed 2 chips using yakara's pinout, have double checked the connections with a multimeter and everything is identical on both boards however only one board can make the button combination for the ZR button (which should be START + R)

also it appears that the home and zr button are always "pressed" in the controller that can't register the combo

check this video of the problem
https://streamable.com/uj64ty

and here's why I say that the home and ZR buttons are always pressed
notice how when i turn on the controller the ZR is pressed and the home "long press" menu pops up
https://streamable.com/77klzu

victorkte, 2023-07-09:
@eduardocintas did you have this problem?

JPZV, 2023-07-09:
Hi, I don't have a N64 around here, I'm waiting for my supplier to bring me two so I can make more tests.

How do you make the combo? First press and **release** Start then press and release R, or do you press and **hold** Start then press and release R?

Meanwhile, I'll try to compile a new binary so you can test it. I may found the glitch, but, again, I don't have the hardware here.

JPZV, 2023-07-09:
@victorkte Please try this binary: 
[BluN64-Yakara.zip](https://github.com/JPZV/BluN64-ESP32/files/11995320/BluN64-Yakara.zip)

You have to unzip it and flash it from 0x0 (i.e. set the address to 0x0 on your ESP Flash Tool)

victorkte, 2023-07-10:
@JPZV

It fixed the start + R combination, it now registers ZR properly but it's a bit hard to pair (i'm pairing directly from the test board by the way) 

also it changes the pinout to your pinout, i'm using yakara's pinout (without yakara's pinout is a bit hard to activate thebuttons directly from the board so i couldn't confirm if  start + L was also fixed

here's the log when it gets stuck pairing

W (57656) BT_HCI: hcif disc complete: hdl 0x81, rsn 0x13
I (57656) ns_bt_gap_cb: UNKNOWN GAP EVT: 17
W (59972) BT_HCI: hcif conn complete: hdl 0x81, st 0x0
W (59983) BT_HCI: hcif link supv_to changed: hdl 0x81, supv_to 8000
I (59993) ns_bt_gap_cb: UNKNOWN GAP EVT: 16
I (60122) ns_bt_gap_cb: authentication success: Nintendo Switch
I (60141) ns_bt_gap_cb: 58 2f 40 27 0c 31 
BlueN64 Control Event: 
	type: 0
	event: 4
	param: 0
W (60294) BT_HCI: hcif disc complete: hdl 0x81, rsn 0x13
I (60295) ns_bt_gap_cb: UNKNOWN GAP EVT: 17
W (62563) BT_HCI: hcif conn complete: hdl 0x81, st 0x0
W (62572) BT_HCI: hcif link supv_to changed: hdl 0x81, supv_to 8000
I (62582) ns_bt_gap_cb: UNKNOWN GAP EVT: 16
I (62764) ns_bt_gap_cb: authentication success: Nintendo Switch
I (62766) ns_bt_gap_cb: 58 2f 40 27 0c 31 
BlueN64 Control Event: 
	type: 0
	event: 4
	param: 0
W (62864) BT_HCI: hcif disc complete: hdl 0x81, rsn 0x13
I (62865) ns_bt_gap_cb: UNKNOWN GAP EVT: 17
W (63865) BT_HCI: hcif conn complete: hdl 0x81, st 0x0
W (63871) BT_HCI: hcif link supv_to changed: hdl 0x81, supv_to 8000
I (63875) ns_bt_gap_cb: UNKNOWN GAP EVT: 16
I (64058) ns_bt_gap_cb: authentication success: Nintendo Switch
I (64063) ns_bt_gap_cb: 58 2f 40 27 0c 31 
BlueN64 Control Event: 
	type: 0
	event: 4
	param: 0

you don't actually need an n64 controller to test this as it can be tested directly from the dev board, on the dev board  (with the broken build) sometimes it won't pair unless you touch the lower pins (see the video)

and you can test the START + R combination by shorting GND + D15 (super close) and D32 like in this video here

https://streamable.com/1omk8c

victorkte, 2023-07-10:
I also noticed something

the chip that I installed and is working is the same chip from my video review 19/may/2022
[image]

howerver all of my newest chips, none of them work and they're newer chips 11/2023
[image]

Could it be that something was changed?

JPZV, 2023-07-10:
> also it changes the pinout to your pinout, i'm using yakara's pinout

Yeah, I forget to change the pinout while compiling. Please, try this 
[BluN64-Yakara.zip](https://github.com/JPZV/BluN64-ESP32/files/12003064/BluN64-Yakara.zip) version again, now it should work perfectly.

I know I could use a Dev board for testing, but when it comes to specific things like combinations or pinout, I would rather to use a real N64 controller because it's closest to a real environment.

> but it's a bit hard to pair

I would need to check this. Right now, pairing to my Switch is pretty much straight forward. Please, when you can, give me any detail about your Switch, like Hardware model and revision (e.g. Original Switch Rev 1, you can check that with your Serial Number and looking for some info in Google), Software version, and anything that you consider to be useful for this.

> howerver all of my newest chips, none of them work and they're newer chips 11/2023
> 
> Could it be that something was changed?

It may be something different, but it should affect that much.

First thing first, clean up every flux rest on your board and on your controller. It happened to me before that some flux may make a short circuit between the GPIO, so every time I work with any type of flux, I clean up my board with some isopropyl alcohol until it's shinning again.

Also, if you have a multimeter or a voltimeter, check the voltage or the continuity between ground and every button. Those should be 3.3V/open while being unpressed, and 0V/close when being pressed. If you have floating values (i.e. random voltage in either state off from the 5% tolerance) or the wrong voltage from either state, then you may have a wiring problem.

victorkte, 2023-07-11:
> Please, try this
tried the new bin it has yakara's pinout but now the button combination doesn't work anymore

> Please, when you can, give me any detail about your Switch, like Hardware model and revision

my switch is a patched V1 in the latest version 16.0.3
[image]

>First thing first, clean up every flux rest on your board and on your controller. It happened to me before that some flux may make a short circuit between the GPIO, so every time I work with any type of flux, I clean up my board with some isopropyl alcohol until it's shinning again.

actually the new chip installation is as clean as possible, while the old chip is bathed in flux and full of floating values lol, 

here's some high detailed pictures and videos
NEW CHIP (pairing difficulties/button combo issues)
[image]
Testing voltages (tested them directly in the esp32 pads too, they're the same there)
https://streamable.com/9hir84

OLD CHIP (Working perfectly)
[image]

https://streamable.com/54cnpf

victorkte, 2023-07-11:
Also LED behaviour is different on the old chip it starts blue then switches to red

but on the new chip it already starts red, the led is different on the video but its wired identically and i have a 3rd board with a new chip soldered with the exact same led and it behaves just like that

https://streamable.com/lwa86x

is there a tool for recovering the flashed firmware? could it be that yakara's website is taking files directly from your repository and when you made that first change trying to fix the Z button being registered as L2&R2 it messed something up?

victorkte, 2023-07-12:
@JPZV any updates on this?

I would appreciate if you could fix this because I'm about to make the second version of my custom n64 pcb it uses yakara's pinout so if i have to change to your pinout I'll have to do it before ordering
[image]

JPZV, 2023-07-12:
Hi, sorry, yesterday was a pretty busy day for me. Today I'll investigate about this issue. I may have an idea about what could be wrong.

By the way, nice PCB! It looks amazing!

JPZV, 2023-07-12:
@victorkte I think I fixed the combo issue. Please try this new 
[firmware](https://github.com/JPZV/BluN64-ESP32/files/12031199/BluN64-Yakara.zip) and let me know if it's fixed now so I can push the changes and create a new version

victorkte, 2023-07-13:
> Hi, sorry, yesterday was a pretty busy day for me. Today I'll investigate about this issue. I may have an idea about what could be wrong.
> 
> By the way, nice PCB! It looks amazing!
> 
> I'm so glad that you use our project for that. Just keep in mind that this project and HOJA uses a non-commercial license, so you may have to ask permission to both me (which I don't have any problem) and [Mitch](https://github.com/mitchellcairns) if you want to sell it as a final product

I actually made it to showcase on my channel, I don't plan to sell it as making custom PCBs is not that cheap compared to large scale manufacturing

I tested the firmware build on the dev board and its working this time!, I'm gonna solder it and test if it can switch to bluretro mode on the controller later

victorkte, 2023-07-14:
@JPZV It's almost perfect now!

All the buttons are working flawlessly and the button combo works as well!

there is just 2 things to fix 

1 the analog stick is not being detected at all (maybe the analog part is still configured to your pinout and not yakara's?)
2 the button combo to change to Blueretro mode is not working as well (L+R+START for 7 seconds) the LED didn't change colors too

Please take a look at it when possible
I'll send you the custom PCB files to you as thanks for going through this hassle =)

JPZV, 2023-07-14:
> 1 the analog stick is not being detected at all

The N64 controller's joystick is a really mess to work with. I hope the following bin fixes that issue. Just for the record, you're using an original joystick/analog, right?

> 2 the button combo to change to Blueretro mode is not working as well

I think I know what is the problem. It's fixed on the following bin too, but if you want to also check, try to hold L+Dpad Down+Dpad Up (you may need to short the GPIO for one of the DPAD). If that changes the mode, then there's the problem.

Please, try this 
[binary](https://github.com/JPZV/BluN64-ESP32/files/12053681/BluN64-Yakara.zip) and let me know if that fixes both problems.

victorkte, 2023-07-14:
Yes I'm using original sticks restored with parts from kitsch bent , I'm gonna install it later and report back

victorkte, 2023-07-15:
@JPZV the buttons are messed up now, I think you accidentally changed to your pinout again instead of yakara's

JPZV, 2023-07-16:
You're right! I forget to set the pinout to Yakara's before building.

Please try this [firmware](https://github.com/JPZV/BluN64-ESP32/files/12063090/BluN64-Yakara.zip) instead

victorkte, 2023-07-16:
Ok reflashed it and here's the feedback

1 button combo now changes to android mode correctly but on android the C buttons and the analog stick are not working (Z button registers as just one trigger which is great!)

2 Led won't change color when using bluN64 mode

3 Analog stick is still not working on switch mode too

maybe the analog stick pinout is configured to your pinout instead of yakara's?
remember that the stick on yakara's pinout is on a different location too so it's not just changing the buttons to yakara's you have to change the analog stick pinout too

victorkte, 2023-07-16:
do you want to see how the buttons are configured on android side on the other working controller?

JPZV, 2023-07-16:
> Ok reflashed it and here's the feedback
>
> 1 button combo now changes to android mode correctly but on android the C buttons and the analog stick are not working (Z button registers as just one [...]

Ok, definitely I'll need to get a N64 controller to test it. Tomorrow I'll try to replicate an analog stick using a protoboard or some alternative analogs that I have around here, but if I couldn't, then we must wait until my supplier get me a brand new N64 controller 

> maybe the analog stick pinout is configured to your pinout instead of yakara's? remember that the stick on yakara's pinout is on a different location too so it's not just changing the buttons to yakara's you have to change the analog stick pinout too

Nope, it uses Yakara's. You can check it on [n64-pad.h#L72](https://github.com/JPZV/BluN64-ESP32/blob/main/n64-sdk/n64-components/include/n64-pad.h#L72). Yesterday I double checked it and the binary that I uploaded before I swiped between the _Q and the _INT for each axis, but it seems it didn't do anything

victorkte, 2023-07-16:
Is it easy to dump from the esp chip? I could try dumping from the working controller to see if there is any differences

victorkte, 2023-07-16:
@JPZV I made a discovery here

can you try removing all the updates from the online installer here https://jpzv.github.io/BluN64-ESP32/ and leave only the 2023.1.1?

I was checking the video again to see if I could spot something different and yakara's installer may be getting the updated files from your repository and making changes to it 
this is how the install window looked like on the video 
[image]

and now when I install from his online installer it shows this 
[image]
notice that it's installing JPZV-2 so definetly its a problem with the newer updates

victorkte, 2023-07-16:
also there is something on the logs now that doesn't appear in the logs on the video

>W (466) rmt(legacy): legacy driver is deprecated, please migrate to `driver/rmt_tx.h` and/or `driver/rmt_rx.h`

perhaps its indeed fault of the modern version of this chip

victorkte, 2023-07-17:
@JPZV any success?

JPZV, 2023-07-17:
Nope, and I have talked to Yakara yesterday because I want to release the new project called BluControl which will replace the BluN64 project because it's more robust by a lot, and there shouldn't be problems like this. But first, Yakara has to release some videos and then we can release the new project.

Aside of that, I got a new N64 Controller which will be delivered to me between this Friday and next Monday. With that I can test both BluN64 and BluControl properly and fix any issue without having to give you the firmware with any minor change.

Whatever it comes first, will give you any update asap, and hopefully a new update. I'll anyway pull the last commit which will create a new binary from GitHub Actions, but I'll not upload it to the web installer as it still has some issues as you pointed out

victorkte, 2023-07-17:
That's cool!

this blucontrol has rumble support too right?
can't wait to test it out!

JPZV, 2023-07-17:
Can't confirm nor deny before releasing it, but it seems you know the answer ;)

victorkte, 2023-07-17:
how many videos yakara has to release before you can publish the project?

victorkte, 2023-07-17:
I hope it doesn't take too long, because I was modding 2 more controllers for a friend =/

victorkte, 2023-07-17:
>Whatever it comes first, will give you any update asap, and hopefully a new update. I'll anyway pull the last commit which will create a new binary from GitHub Actions, but I'll not upload it to the web installer as it still has some issues as you pointed out

is it too hard to pull the versions from the web installler? and leave only 2023.1.1?

JPZV, 2023-07-17:
I'll hide the builds for the Yakara's pinout, but I'll leave the others version as those have some bug fixes

victorkte, 2023-07-17:
I asked because I wanted to check if changing that would change the installer on yakara's page as it seems to be pulling from your repository so it will always get the most recent version

but it's ok if you can't, I'll wait for the blu control project

JPZV, 2023-07-17:
Because of how both GitHub Pages and ESP Web Installer work, his installer and mine are completely different and work by their own. Even the binaries are different, because mine are just one generated by GitHub Actions, while Yakara's are separated binaries (bootloader.bin, ota.bin, switch.bin, blueretro.bin, etc.) and all made by hand.

Anyway, I'll leave the Yakara's binaries from my installer as hidden, because of the pinout issue.

I'll update you as soon as I get any new information

victorkte, 2023-07-22:
@JPZV 

I decided to re-check all connections and actually I had forgotten to feed 3.3v to the analog stick 😝
now the analog stick works perfectly, the only thing not working is the LED color change, but that may be because I'm using a diffferent LED for this build so it might be a faulty LED in the end

sorry for the trouble

but I'm still looking forward to blucontrol, I'll make changes to my custom PCB to be able to solder a rumble motor and make it even better =D

JPZV, 2023-07-22:
Nice! It happens a lot actually. For example, yesterday I send to PCBWay a PCB and today I realized I forget to connect the 3.3V of the ESP32 to the power node. I'll have to add a wire when I get my PCBs 🙃

Could you confirm which things are not working then? Aside of the Lights, which I can confirm it's not your fault, it's actually confirmed that there's an issue on my code. I'll try to fix it now

victorkte, 2023-07-22:
Only the LED

The button combo works
And its changing to bluen64 perfectly too

JPZV, 2023-07-22:
Ok, please, try this 
[binary](https://github.com/JPZV/BluN64-ESP32/files/12136849/BluN64-Yakara-ESP32.zip). It was compiled using GitHub Actions, so if it's working fine, then I'll upload it as an Update

victorkte, 2023-07-23:
Nope, the LED stays the same and the analog stick is reversed now

JPZV, 2023-07-23:
About the analog, I was expecting something like that, as I wanted to verify something about the pinout. I'll change it on the next commit then.

About the LED: That's strange. Are you sure the connections are right? Are you using a Common Cathode RGB, like in [Yakara's schematic](https://blogger.googleusercontent.com/img/b/R29vZ2xl/AVvXsEjYVVNazktb112lT7u4XjNIBYx66Q7ZrU1JEnDORoSN_dr5oF6VwJjzFlBq_9--vRPlnLuCCC0ZeR-Q2l0XOHA-yxPT9OLS_jmQW66kpnmaIo6EsBVNxMCNrLW_vvZOxvqQmX4dy-1L9-oVNl9p_gK25WHH4c-sWkrviOV0bqO5rNVDqb-AGMWPLQM97w/s3508/n64%20universal.png), right?

victorkte, 2023-07-24:
yes I checked with a multimeter, also there is no voltage coming from pin IO4 when in bluen64 mode

I'm gonna deliver the controllers to my friend tomorrow can you make the changes soonish?
because after that I'll take a break from messing with these boards (I'll come back later to fix my main controller and modify the custom pcb)

victorkte, 2023-07-24:
thought the  LED is common cathode it is just a red/green LED I'm not sure if this makes any difference

JPZV, 2023-07-24:
I'll try my best.

The joystick issue is not a problem, it's actually fixed and it'll available in the next update.

The thing is the LED.

Can you confirm the LED is connected like this:

```
Cathode -> Resistor -> GND
Anode 1 -> 3.3v
Anode 2 -> GPIO4
Anode 3 (If aplicable) -> Not Connected or GND
```

> I'm not sure if this makes any difference

it shouldn't. The important thing is that the GPIO4 should be at 0.0V when on Switch Mode, and ~3.3V while on Bluetooth (AKA BlueRetro) mode.

JPZV, 2023-07-24:
By the way, please, try with this [build](https://github.com/JPZV/BluN64-ESP32/files/12141067/BluN64-Yakara.zip). I made it manually using the same command lines as in GitHub Actions, though the checksum are different for some reason (maybe the compilation timestamp?). It may have the analog issue, but it should have the LED fixed (I hope)

victorkte, 2023-07-24:
Exactly I just opened the other controller "the one with the working led" and confirmed it's 3.3V on Switch Mode, and 0.0V while on Bluetooth

but I didn't had voltage on none of the modes, 

I'll flash this new build and report back

victorkte, 2023-07-24:
@JPZV yes the LED was fixed in the last build but as you said the analog stick is still reversed

can this be fixed?

JPZV, 2023-07-24:
It should be [now](https://github.com/JPZV/BluN64-ESP32/files/12141334/BluN64-Yakara.zip). I pushed the fix, but it needs to be compiled with GitHub Action, which could take up to 20 minutes. I compiled it manually, so please, test it again with this compilation while I wait for the final to be compiled so I can compare both checksum.

victorkte, 2023-07-24:
I'd rather wait for the final compilation dessoldering the wires to flash the chip is very annoying, i'll wait more 30 mins and get the final compilation

JPZV, 2023-07-24:
Ok, that's completely understandable, that's why I always leave some headers for serial (UART) connection even on my PCBs.

Here's the [final](https://github.com/JPZV/BluN64-ESP32/files/12141452/BluN64-Yakara-ESP32.zip), compiled directly from [GitHub Action](https://github.com/JPZV/BluN64-ESP32/actions/runs/5640481816). I'll cross the fingers so it's working correctly now 🙏🙏🙏🙏

victorkte, 2023-07-24:
@JPZV now L doesn't work and start is triggering both start+L and thus bringing up the home menu

why the button configuration gets messed up like this? do you have to code everything by hand everytime or you can just copy and paste?

I checked the wirings too and there is no shorts

JPZV, 2023-07-24:
Okey, that's completely weird, that shouldn't happen for whatever reason. Anyway, looking at the GitHub Action, I found out that I had a Typo, defining one of the Combo GPIO to 35, instead of 15. 35 is being used for the Home Button, which is not in use anymore and it's there just for Forked projects. But that doesn't explain why you're getting pressed both L and Home.

Answering your question: This project was one of my first made with the ESP-IDF framework, and it has maaaaaaaaaaany issues related of how it's worked. One of them being the highly reliably on header files instead of using the sdkconfig. This means that every change must be done practically on-code. Meanwhile, BluControl was make almost from the ground, with keeping in mind about the people who don't know how to even modify a header file without messing up. For this, I use only sdkconfig for any type of modification.

As BluN64 and BluControl share many thing, I decided to use BluControl-SDK project as an intermediary, so I can keep BluN64 up to date while working with BluControl. Sadly, as I said before, BluControl uses only sdkconfig, and both the LED GPIO and the Mode Switching GPIOs are defined there. So, I had to use the `sed` command from Linux to replace the GPIOs from the original pinout to the new GPIOs.

Luckily, this shouldn't happen at all on BluControl, because every type of pre-defined controller will have its own sdkconfig

Anyway, here's the final final final final compilation:

victorkte, 2023-07-24:
Nothing changed, L still doesn't register and start is still as the combo,

and honestly at this point I give up, I even burned my fingertips while soldering, I'm gonna flash the build without the LED and call it a day, I'll come back when blucontrol is released

victorkte, 2023-07-24:
Ok now something is wrong on my side 
I flashed that other build without the LED working and now start is still working as the button combo????

I double checked with the multimeter and there are no bridges on the soldering points could it be a faulty installation?

victorkte, 2023-07-24:
@JPZV ok now its working perfectly

turns out there was a tiny bridge from ground to L that somehow I didnt detect at the first time (probably because my chip is soaked in flux by now)

I reflashed it with this version

> Here's the [final](https://github.com/JPZV/BluN64-ESP32/files/12141452/BluN64-Yakara-ESP32.zip), compiled directly from [GitHub Action](https://github.com/JPZV/BluN64-ESP32/actions/runs/5640481816). I'll cross the fingers so it's working correctly now 🙏🙏🙏🙏

and now its working perfectly, LED, button combo, bluretro (even tested it on my console)

can you please ask yakara to update his web installer with this? because I made my entire tutorial using his installer page

JPZV, 2023-07-24:
That's a good news!

Could you please confirm that the [Final version](https://github.com/JPZV/BluN64-ESP32/files/12141798/BluN64-Yakara-ESP32.zip) if working properly? Because it has the typo fixed. In theory, without that fix, you shouldn't be able to switch between the Switch mode and the Bluetooth one.

In any way, I'll let him know :)

victorkte, 2023-07-25:
Oh I already delivered the controllers, but don't worry I tested everything and my friend also tested on his switch & smartphone and everything is working perfectly

So if it isn't broken don't fix it 😂

JPZV, 2023-08-09:
I'll close this as completed. Feel free to re-open if you find a related issue or open a new issue as you like

closed 2023-08-09 by JPZV