JPZV/BluN64-ESP32 #9: Physical buttons for Home, Capture and Turbo

vitorpradobr, 2023-08-30:
I'm making an arcade controller for NS following Yakara's guide. 

I would like to add physical buttons for Home, Capture and Turbo.  Can you help me? 

I don't know how to program, so I'm very grateful to Yakara's guide and everyone involved.

vitorpradobr, 2023-08-30:
Holy, that's exciting! Hope this project come soon, I'll be waiting.

Meanwhile, is it possible to add just the Home button or at least a shortcut like Select+L=Capture / Select+R=Home?
Adding a 10k resistor isn't a problem.

Where could I ask for this? I'm using the firmware version of this guide "https://yakaracolombia.github.io/esp32-online-tool/arcade2.html".

JPZV, 2023-08-31:
If I recall correctly, that controller uses BluControl (every controller made by Yakara after the GCN one uses BluControl). As BluControl is fully customizable, I don't have any control about making the combo available. Sadly I watched the entire video and he didn't mention anything regarding the Switch functionality. I'm trying to asking him directly but still no response yet about the combination.

BTW: DON'T update/flash your ESP32 with any version from this project (BluN64), because this was made specifically for the N64 controller, and it will break almost everything

TheIcedTeaAddict, 2023-09-07:
> For N64 controllers, the Home button was enabled in GPIO 02 (or GPIO 35 if you're using Yakara's pinout) up to the v2023.04.2, then it was disabled on v2023.07.1 due to compatibility issues, specially when using Yakara's pinout because GPIO35 doesn't have an internal Pull-down resistor, so you have to add it yourself while soldering (i.e. you have to add an 10K resistor between the GPIO and ground, along with the wire to the physical button).
> 
> About the Capture Button, I think it was never implemented, I don't remember why, but I think it's because we don't have any [GPIO available](https://github.com/JPZV/BluN64-ESP32/blob/main/hardware/ESP32%20Pinout.png) for extra features/buttons. We could use GPIO00, but it can be easily miss-pressed while powering on the controller and entering to Download-Mode by accident.
> 
> About Turbo buttons, I actually have never used that feature in any controller or emulator on my life, so I never had an idea of implementing it. I'm not sure if it's that important, and if it's, then we'll have the same issue with the GPIOs availability.
> 
> In any way, BluControl, my yet-for-release project about DIY BT Controllers, has both Home and Capture buttons available so you can build your own controller without knowing how to program. You'll just need to follow the _Build your own_ guide, download VSCode, make some clicks, and flashing it to the controller. Sadly, I'm still waiting for Yakara giving me the green light so I can release that project

Hey, sorry for trying to contact you in this way, but I honestly have no idea how to DM on Github for the life of me, and I felt this topic was fitting enough for now. 
I'm super excited to see BluControl and all the possibilities that can be made with it. However, I have seen that you've mentioned the issue of running out of GPIO pins and being unable to implementing some features. Is there anything that prevents the use of Shift Registers for those who are willing to use them? 

If I recall they can be used to add GPIOs for digital signals such as buttons and LEDs, even with the possibility of Turbo and reactive LEDs, thus leaving several GPIO pins for other features, like analog, rumble (if it is not possible with the shift registers for whatever reason), motion control, charge/usb connection detection? (I know the TP4056 module does not support charge and play, but there are a few modules that do if I recall correctly. MCP73871 modules might be able to.) NFC for Amiibo? Maybe even a switch that can change an arcade stick's up, down, left and right between the dpad, left stick and right stick, similarly to Brook Fighting Boards. (Again, if that is not already possible with the Shift Registers.) And pretty much any other feature a controller could want. 

I have made an image of how the Shift Registers can be hooked up to the ESP32 just to help get the idea. I also believe the theoretical limit of how many Shift Registers can be daisy-chained is 8, so that does make for a lot of possible digital GPIOs, about 64, pretty much overkill for any project, haha.
[image]
This example has three 74HC595 Shift Registers connected to the ESP32.

If there is a better way to contact you, let me know please and thanks. Thanks for your work on BluControl, and I hope this suggestion helps in any way. Sorry again contacting here and for the long write, haha.

JPZV, 2023-09-08:
Sorry for the late answer

@TheIcedTeaAddict that's actually a very good idea, and funny enough, that's a planned feature for BluControl. It's not fully implemented yet as I need to design a nice way to make it configurable without having like 100 fields on menuconfig, but I'll definitely add it to BluControl.

While for BluN64, I don't think I'll do it here, as the project is entering in a LTS state, without adding major features and just fixing bugs/glitches.

About how contacting me: Sadly GitHub doesn't have DMs, and my email is for personal use only. But if you want, you can contact me on Twitter (I refuse to call it X 🤮). I'm not that active there, and most of my post are on Spanish, but I'd be glad to answer any question by DM. My Twitter ID is [JuanpZV](https://twitter.com/JuanpZV)

TheIcedTeaAddict, 2023-09-08:
No worries about the late answer.

Sweet, glad to know that it'll be implemented into BluControl. As for how to configure it in a nice way, I'll be honest, I'm not currently sure how you're planning to make it configurable at the moment even with the mention of menuconfig, I'm not familiar with these terms, but maybe there could be a online or offline (preferably) GUI configurator program? If it is not too much to put together of course. It could be used by the user to select features to be used, number of Shift Registers and assign functions to compatible GPIO pins on the ESP32 and the selected number of Shift Registers. Probably would be the cleanest way to configure it.

Fair enough about BluN64, I assume it could possibly be remade though BluControl for any features included others may want regardless. 

Haha, fair enough about that too, the website itself barely knows what it is even called anymore anyways. I checked out your Twitter but it seems like your DMs aren't open. I have followed you and Yakara however. I'll probably ask about any other questions, make possible suggestions and try to even lend a hand when I can, when I am able to get in contact on there. My Twitter ID is [TheITAddict](https://twitter.com/TheITAddict)