JPZV/BluN64-ESP32 #13: Gamecube analog stops working

curcius, 2024-04-10:
Hello everyone, my thanks to the developer of this incredible project.
 I modified my controller a few months ago but I was very dissatisfied with the stick, so I decided to buy a gamecube style stick like the photo, I replaced it but in blueretro mode it works for a few minutes and suddenly it stops working for some sides mainly downwards, the other sides continue to work. 

I thought it could be a defect in the stick, but in Nintendo Switch mode it works perfectly for hours without any defect. 

Could anyone give me an idea of what might be happening? Thank you very much. 

[image]

curcius, 2024-04-10:
I made a video. I forgot to inform you that my blueretro version was 1.9.2, I went back to 1.9.1 but the same problem continued unfortunately. 

https://youtu.be/oamgQhMjGbg

JPZV, 2024-04-10:
Hi! Thanks for testing this project! I know it has been a while, and sorry that you're having issues.

I always recommend to use the originals one if possible. But, in any case, both in BlueRetro and Switch mode the BluN64 uses the exactly same code for obtaining the Joystick position.

I can think in two possible reasons:

1) There's a timing issue. Because the GC style joystick has a IC to emulate a N64 joystick, and then the BluN64 tries to interoretate the signal from a real N64 joystick into an analog joystick, there may be a mismatch somewhere. To test this is pretty simple: Just try to use the original Joystick. Even if it has the loose stick issue, if it works 10/10 everytime, then it means that the GC one is not that straight forward with BluN64 and some improvement has to be done in our project.

2) Maybe BlueRetro made a change since the last time I tested this project (to be honest, I was very busy this year. In no means I wanted to abandon nor archive this project, but I have almost no free time to work with it). To prove this is the same as 1). If the issue persists even with the original Joystick, then there's an issue with the communication between BluN64 and BlueRetro

I think I have one of those joystick somewhere, and I have my N64 controller and my N64 near me. The problem is that I have almost no time until the next next week (or even until May). If you, by that time, can bring me the most information that you can, like logs, workaround, findings, testings, etc. I may be able to start working in a fix knowing where should I start and we can maybe have a new update asap.

curcius, 2024-04-10:
> Hi! Thanks for testing this project! I know it has been a while, and sorry that you're having issues.
>
> I always recommend to use the originals one if possible. But, in any case, both in BlueRetro [...]

@JPZV Thank you very much for responding so quickly, it really is something intriguing because in switch mode it works perfectly. Regarding the original Switch stick, I can tell you that I haven't had any problems since I modified the original control. I'm going to learn how to collect the logs to try to identify this failure, I was thinking that it could be something related to the 3v power supply, but if it were something like that the same failure would occur on the switch. I really don't know what to think anymore, I think only a log can clarify this for us. I will study the case with more information. Thanks.

curcius, 2024-04-10:
@JPZV I forgot to mention that my 64 with blueretro works perfectly with other controls, including generic bluetooth controls from aliexpress. 

This for example:
[image]

curcius, 2024-04-14:
Unfortunately I couldn't discover the fault and had to give up. I'm going to do another project that uses a common stick with potentiometers so I thought about using that same stick so I opened it to check and it actually has a shaved chip with no reference, probably an FPGA. 

I'm going to use the xgames david project to be able to use this stick which is many times better than the original n64. I kept the chip and if one day @JPZV  finds time to update the project to use common analogues I'll update my controller again. 

Follow the photos of this Chinese stick 

[image]

[image]

This is the modification I made to be able to use it with the other project: 

[image]

curcius, 2024-04-19:
I thank everyone @JPZV for their help, I managed to improve the situation a little because I realized that if I turn it off and on several times the control stabilizes the stick commands, it stays for a long time without locking, but I believe it is some error in blueretro's interpretation because so much in the Nintendo Switch and other devices I don't notice the error. 

I'm going to open an issue on blueretro's github to see if anyone has any insight into a definitive solution. Thank you again for this wonderful project and for the attention given.

curcius, 2024-04-19:
This is my contact if one day you return to work on the project, it would be a pleasure to help. 

andre.curcio@gmail.com

curcius, 2024-04-20:
Finally I managed to solve it, I don't know why but the configuration that worked is the one in the photo:

[image]

Thank you very much for your help and I'm very happy that it works now =)

curcius, 2024-04-21:
.

closed 2024-04-21 by curcius